victronenergy/venus-html5-app #459: Bug: Layout on Garmin 9000 needs love...

mman, 2024-10-21:
Everything way too small...

[image]

mman, 2024-10-23:
Looking at the specs for Garmin 9000 (for example here: https://panbo.com/garmin-gpsmap-9000-series-high-resolution-and-big-power/) it looks like it's a 27" screen with full 4K (3840 × 2160) resolution rendering 1:1.

I was able to simulate the layout on my 27" 5K by turning off the Retina and then we get layout very similar that you can see on the picture above, so I think I now have a test bed to investigate the necessary CSS.

[image]

mpvader, 2024-10-23:
👍

mman, 2024-11-08:
Displays come with variety of resolutions:

Let's say that up to the resolution of HD (1280x720) and Full HD (1920x1080) the displays are typically used in 1:1 mapping, where one point on the screen is rendered as one pixel. HTML by default uses `font-size: 16px` for these.

When you double the HD, and Full HD resolutions, you get 2560x1440 (2K) and 3840x2160 (4K).

When these screens are used for example on a TV, they still render 1 device point as 1 pixel and report full resolution in pixels.

When they are used on a laptop or monitor, they may report the resolution in points as HD, or Full HD, and render 1 point into 4 pixels - hiding the physical resolution, but indicating the to the browser via `window.devicePixelRatio = 2` that it is a "retina" screen with higher DPI. The browses in such cases still uses `font-size: 16px` but because it knows that `window.devicePixelRatio == 2` it draws a text that is in the end `32px` tall.

These are in short called High DPI screens. It's the responsibility of the operating system to properly set `window.devicePixelRatio` for these High DPI screens depending on the use. So TV will have `1` but laptop should get `2`.

Looks like Garmin 9000 series is simply picking up full 4K screen and reporting it as `devicePixelRatio = 1`. In their own words this means that they essentially provide 4 small Full HD screens in one glass for the 4 apps to use.

I have addressed this issue with two commits: 

https://github.com/victronenergy/venus-html5-app/commit/2f09cd84a1309f8bd21ffa17d8516c0ee7a9c81d increases html base font size in pixels when it detects that we are on a screen that reports > 1920px and > 2560px in horizontal resolution.

https://github.com/victronenergy/venus-html5-app/commit/c78e3dcd6eeb5a0d483804c3a86f46eafc9ea782 reports `devicePixelRatio` in diagnostics so that we can confirm on real new devices with such screen how is the browser configured.

More work is possibly needed on other CSS tweaks (see for example the too long battery name in the screenshot below) but for now these two commits make the app layout behaviour on Garmin 9000 work as if it was a more traditional device with Full HD screen.

[image]

closed 2024-11-08 by mman