victronenergy/venus-html5-app #477: Generator data not shown, which is present in gui-v2

martinbosma, 2025-04-23:
From community: https://community.victronenergy.com/t/victron-energy-garmin-onehelm-generator-running-but-not-displaying-gen-loads-after-venus-v3-54-upg/26524

[image]
[image]

System:
- Quattro 12/5000/120-2x100 120V in split phase (L1-L2)
- AC Input 1 as generator
- AC Input 2 as shore
- Relay 1 function: Generator start/stop
- Phase count: 2
- System/VebusInstance: 276
- Vebus/DeviceInstance: 276

mman, 2025-04-25:
This is caused by the fact that we detect a relay controlled generator, we display it, but we never properly set its active state and so all values remain hidden.  Key logic begins at line 193.

https://github.com/victronenergy/venus-html5-app/blob/3d27275e2288c793ff02ce68fdbc6b3527dae204/src/app/Marine2/components/boxes/DevicesOverview/DevicesOverview.tsx#L186-L220

This is despite the fact that generator itself is properly detected as running which is done by examining both the `active` property as well as the `statusCode` property

https://github.com/victronenergy/venus-html5-app/blob/3d27275e2288c793ff02ce68fdbc6b3527dae204/src/app/Marine2/utils/formatters/devices/generator-relay/generator-state-for.ts#L4-L17

mman, 2025-04-25:
Looks like this may be very simple fix:

```diff
--- a/src/app/Marine2/components/boxes/GeneratorRelay/GeneratorRelay.tsx
+++ b/src/app/Marine2/components/boxes/GeneratorRelay/GeneratorRelay.tsx
@@ -43,7 +43,7 @@ const GeneratorRelay = ({
   const { electricalPowerIndicator } = useAppStore()
   const { current, voltage, power } = useActiveInValues()
   const phasesData = usePhasesData(phases, voltage, current, power)
-  const status = active ? "active" : "inactive"
+  const status = active || statusCode === 1 ? "active" : "inactive"
 
   if (componentMode === "compact" && compactBoxSize) {
     return (
```

mman, 2025-04-25:
Yet better fix may be to consider:

1. If Venus is configured with relay controlled generator
2. And if we know the `statusCode` of the generator
3. We may assume that generator is `active` when `statusCode == 1`

```diff
--- a/src/app/Marine2/components/boxes/DevicesOverview/DevicesOverview.tsx
+++ b/src/app/Marine2/components/boxes/DevicesOverview/DevicesOverview.tsx
@@ -193,6 +193,7 @@ const getAvailableDeviceBoxes = function (
         <GeneratorRelay
           key="generator_relay"
           {...generatorRelay}
+          active={generatorRelay.statusCode === 1}
           componentMode={componentMode}
           compactBoxSize={compactBoxSize}
         />
```

This fix has a slight drawback that can be described like this:

Quote from Quattro documentation:
> Important features:
Two AC inputs; integrated switch-over system between shore voltage and generating set
The inverter/charger features two AC inputs (AC-in-1 and AC-in-2) for connecting two independent voltage sources. For example, two generator sets, or a mains supply and a generator set. The inverter/charger automatically selects the input where voltage is present.
If voltage is present on both inputs, the inverter/charger selects the AC-in-1 input, to which normally the generating set is connected.

So assuming we ignore the documentation, and connect a generator to AC Input 2, and Shore to AC Input 1, and then start the generator... this fix above will lead to Generator page displaying incorrect values from shore input.

Given that this is describing a scenario where things are "not wired according to the docs... or recommended practice" ... I think it's a reasonable fix...

closed 2025-05-01 by mman