victronenergy/venus-html5-app #493: Feat: Remote console color mode

mr-manuel, 2025-07-31:
On most plotters the system color scheme is light and therefore, when opening the Remote Console with dark theme enabled, the display flashes shortly light, since the HTML page follows the system color scheme.

[image]

I made a few changes for GUIv2 (https://github.com/victronenergy/gui-v2/pull/2398), which will be in the next release, that allows to pass the used color scheme. Add the new query parameter to the existing Remote Console URL so that the matching color scheme is applied:

- HTML5 `Light`: `/?fullscreen&colorScheme=light`
- HTML5 `Dark`: `/?fullscreen&colorScheme=dark`
- HTML5 `Auto`: `/?fullscreen&colorScheme=auto`
- HTML5 `Night`: `/?fullscreen&colorScheme=dark`

Here a sample of the last addition https://github.com/victronenergy/venus-html5-app/commit/2769b8c77ad18e3635a3b91969929dd616ee8085

mr-manuel, 2025-08-26:
The command line options are avalable since Venus OS v3.70~27.

mman, 2025-08-26:
This has been fixed in https://github.com/victronenergy/venus-html5-app/commit/15bbf597e890334ba8a14a17f2c46db4d31f8eaa and GUI-v2 now properly inherits the color scheme from the app, as well as it now displays the "Forced by app to {dark|light|auto}" in Settings/General/Display & Appearance when started from the html-5 app remote console.

closed 2025-08-26 by mman